arduino12/rdm6300 #1: Serial Monitor Issue

Skunki, 2019-02-08:
hi, there. Thank you  very much for sharing your code, this is just what I needed.

Unfortunately, when I start the program, only strange characters appear on my serial monitor (like this: $⸮⸮⸮⸮).

It seems that that the leds are working fine, first the TX led and 13LED turn ON, then TX one turn off and the LED one stays until I remove the transpoder.  But every time it reads another, just another ⸮ appears in my monitor.

Any idea of what is happening?

Thank you again

arduino12, 2019-02-08:
Hi,

Note that the [read_to_serial](https://github.com/arduino12/rdm6300/blob/3b8fc3ea8fc5b2b952e99441e1ae2af2a8ca0ed2/examples/read_to_serial/read_to_serial.ino#L21) example serial baudrate is **115200**bps:
I think your terminal's baudrate is not the same (9600 maybe?)

Good luck..!

Skunki, 2019-02-08:
Yes. It was just that, I changed it before and now it works. Thank you again!

Skunki, 2019-02-08:
close #1

closed 2019-02-08 by Skunki