arduino12/rdm6300 #2: Esp8266 wdt reset

sashahn, 2019-05-06:
Hi, actually my issue is not directly connected with edm6300 library...the issue is if a card is in slot continuously esp8266 will crash- wdt....so, after investing the issue, my conclusion is that Software serial cause it .
What I should do to switch Software serial to Hardware serial? 
OTA is used for firmware upload.
Thank you

arduino12, 2019-05-06:
Hi,

I'll add an optional argument to the constructor so the user can supply his favorite serial device (hardware or software) tonight.

Thanks for noticing..!

sashahn, 2019-05-06:
Thank you for prompt reply.
RDM6300 library works flawless with ESP32-Hardware serial UART2 for 9 days/nights. ID card is in slot almost all the time, few times card is pulled out and back just for tests - no issues appears.
I assume that tag presence for the long time  triggers large burst of interrupts - Software Serial Locks the  esp8266 no time for yielding  - watch dog is not fed  which cause wdt resets.
Oh almost to forgot:
This issue happens only when ESP8266 use WiFI and Serial software.
Best regards

arduino12, 2019-05-08:
Hi,

I have added esp8266 hardware uart support with an [example](https://github.com/arduino12/rdm6300/blob/master/examples/esp8266_hardware_uart/esp8266_hardware_uart.ino)!
Just finish testing with all of my boards: esp32, esp8266, avr..

Let me know if it stops your wdt reset problem :)

closed 2019-05-14 by arduino12