arduino12/rdm6300 #4: Voltage divider is not needed for 3.3V microcontrollers

gutierrezps, 2019-06-05:
I measured the voltage on pins TX and RX of two RDM6300 modules I've got, and they're around 3.3V, not 5V. So the voltage divider is not necessary to connect the RDM6300 to ESP8266.

gutierrezps, 2019-06-06:
Since you're possibly going to update the schematics, I would like to suggest adding the detection LED that can be driven by the module, like this:

[image]

Where the pin numbering goes like this:

[image]

arduino12, 2019-06-09:
Hi,

Sorry for my late reply,
I tested my RDM6300 module as well- and you're right - there's no need for an external resistor divider!

I'd like to attach your schematics, if you can add the following changes:
1. PIN10 isn't a 5V, but a reverse polarity protection via a diode.
2. Rename LED to CARD_DETECTED_LED and add a 1K resistor between 5V and the LED.

Thanks,
Arad

gutierrezps, 2019-06-10:
I think your fritzing schematics are more intuitive for newcomers, but if you really wish to include my schematics:

1. Do you have a source for that? Because in a lot of places is said both pins 4 and 10 are +5V inputs.
2. a) I think "CARD_DETECTED_LED" is very verbose (compared to the other pins), and again this pin is simply known as LED. What I can do is add a bar over the name to indicate it's active low, following [KiCAD Library Convention's Guideline S4.7](http://kicad-pcb.org/libraries/klc/S4.7/). b) Indeed, I was using it without a resistor, but the RDM630 datasheet (the "original" module) uses the pin to drive a PNP transistor.

Sources:
<https://www.itead.cc/wiki/RDM6300>
<http://wiki.seeedstudio.com/125Khz_RFID_module-UART/>
<https://github.com/SeeedDocument/125Khz_RFID_module-UART/raw/master/res/RDM630-Spec.pdf>
<https://datasheetspdf.com/pdf/788733/RDM/RDM630/1>

arduino12, 2019-06-22:
Hi,

OK, let me know about the LED current,
When you're done, please attach the final schematics here and I'll close the issue :)

Arad

closed 2019-08-04 by arduino12