arduino12/rdm6300 #5: No tag detection in certain time frames

gutierrezps, 2019-06-12:
I was trying to use the RDM6300 with the ESP8266, and was facing a weird issue: sometimes the ESP wasn't detecting that a tag was being read. Upon adding some debug code, I noticed the `available()` method from `SoftwareSerial` was returning false, even when there was data being sent constantly to the input pin.

Upon further tests, I found out this problem always happened at a certain time after the last successful SoftwareSerial reading. Then I was able to isolate the problem, make it reproducible, and reported it on plerup/espsoftwareserial#100.

In short, the SoftwareSerial doesn't indicates there's data available between 27 and 53 seconds after the last data received. Not only that, but it seems periodic, i.e. there's a time frame it received data, and then there's a time frame it doesn't receive. That means the reading routines becomes unresponsive in those time frames, like there was no tag being read (even when it does).

It would be great if someone could also try to replicate the problem and report it there on plerup/espsoftwareserial#100.

arduino12, 2019-06-12:
I have added esp8266 **hardware uart** support, see examples..
See #2 - looks similar to this issue.
Let me know if it solves your problem :)

gutierrezps, 2019-06-12:
On issue #2 ESP was having a WDT reset after reading a tag continuously. I am able to read it continuously for many seconds, although I have to admit I'm not using WiFi. My problem is that the ESP doesn't report a tag was read when it is approximated after a certain time being idle. I'm not sure if I'm being clear enough :/

arduino12, 2019-06-24:
[AltSoftSerial](https://github.com/arduino12/rdm6300/tree/master/examples/altsoftserial_based) example solves it!

closed 2019-06-24 by arduino12