arduino12/rdm6300 #6: Add support to AltSofSerial

gutierrezps, 2019-06-15:
Hi again. After facing issue #5 on ESP8266, my workaround was to move the RDM6300 to an Arduino Nano (that I'm already using for other functions) that communicates with the ESP through I2C. And to my surprise, while a tag was being read, the Arduino returned garbage on the I2C, and sometimes even locked the bus by keeping the SDA line low.

So I plugged a logic analyzer and there it was: the I2C transaction was becoming corrupt if data was being sent by the RFID reader.

Here's a proper I2C transaction between the Arduino and the ESP:

[image]

And here's the I2C transaction being corrupted while the RFID is sending the tag data.

[image]

Notice that the I2C transaction is fragmented, and also the bus is locked up after that. After searching about problems with SoftwareSerial, I came across [PaulStoffregen/AltSoftSerial library](https://github.com/PaulStoffregen/AltSoftSerial), replaced it on RDM6300 library, and the I2C problems vanished. The downside was that only pin 8 could be used for Serial RX, but it wasn't a problem for me.

That said, what do you think about adding support to [AltSoftSerial library](https://github.com/PaulStoffregen/AltSoftSerial) on ATmega328 based Arduinos (Uno, Nano, Pro Mini, etc)?

arduino12, 2019-06-22:
Hi,

Amazing work! PulseView is the best!
I didn't see you send a PR for this issue, so I'll try to add AltSoftSerial support in the next week...

Hope [plerup/espsoftwareserial#100](https://github.com/plerup/espsoftwareserial/issues/100) fix will be available soon,
Arad

gutierrezps, 2019-06-22:
Thanks! I've opened the PR #7 right now, take a look and see if it's okay.

closed 2019-06-22 by gutierrezps