arduino12/rdm6300 #10: Use a different pin on Arduino Uno?

Danita, 2020-08-17:
Hi! Thanks for this library, works great with the Uno and RDM6300. I was wondering if there is any way to make it work on other pin than 8 on the Uno, since in my project that pin will be used by the motor shield. I've tried other pins but on the only one it works is on 8. Thanks!!

arduino12, 2020-08-19:
Hi!
I guess the altsoftserial lib uses **[only pin 8](https://github.com/arduino12/rdm6300/blob/master/examples/altsoftserial_based/altsoftserial_based.ino#L11)** on the basic 8bit AVR based Arduinos,
But you can use the example that uses the default software serial lib and **[change here to any pin you want](https://github.com/arduino12/rdm6300/blob/master/examples/read_to_serial/read_to_serial.ino#L13)**.. 

Please tell me if it worked out for you :)

Danita, 2020-08-19:
Thanks! I don't know how I missed that :-) It's working now.

closed 2020-08-19 by Danita