arduino12/rdm6300 #12: Use with Arduino MEGA2560

luxxtek, 2021-05-18:
I successfully used this lib on an Arduino Uno but I cannot seem to get it to work on an Arduino MEGA.
Is it supposed to work with Arduino MEGA too?
Any I/O pin or serial1/2/3 would be fine.

arduino12, 2021-05-18:
**Hi,**

The basic read_to_serial example **[uses pin 4](https://github.com/arduino12/rdm6300/blob/master/examples/read_to_serial/read_to_serial.ino#L13)** as SoftwareSerial RX.

According to **[the doc](https://www.arduino.cc/en/Reference/softwareSerial)**:
_Not all pins on the Mega and Mega 2560 support change interrupts, so only the following can be used for RX:
10, 11, 12, 13, 14, 15, 50, 51, 52, 53, A8 (62), A9 (63), A10 (64), A11 (65), A12 (66), A13 (67), A14 (68), A15 (69)._

So you need to change **[that line](https://github.com/arduino12/rdm6300/blob/master/examples/read_to_serial/read_to_serial.ino#L13)** to something like:
`#define RDM6300_RX_PIN 10`

If you want to use HardwareSerial, replace the **[begin line](https://github.com/arduino12/rdm6300/blob/master/examples/read_to_serial/read_to_serial.ino#L25)** with this:
```
Serial3.begin(RDM6300_BAUDRATE);
rdm6300.begin(&Serial3);
```

**I have just tested both solutions and they all work,
Arad :)**

luxxtek, 2021-05-18:
Not only a nice library but also instant support.
Thanks for this crystal clear clarification.

arduino12, 2021-05-18:
**Happy to help..!**

I have just added **[hardware_uart](https://github.com/arduino12/rdm6300/blob/master/examples/hardware_uart/hardware_uart.ino)** example to support all boards with more then one `Serial`.

closed 2021-05-18 by arduino12

luxxtek, 2021-05-18:
Sorry but I can't seem to get it working on serial1/2/3 with the hardware_uart example.
Does work with one of the mentioned MEGA pins above.

arduino12, 2021-05-18:
Check your wiring..

I connected the rdm6300 module with 20cm wires to:
GND, 5V, mega2560 pin 19 (RX1 pin)
and the hardware_uart example works :)

luxxtek, 2021-05-18:
Sorry for the extra work.
I was mixing rdm6300 module with 7941E and thought they worked identically which is not the case.
I can confirm that it now works OK with rdm6300 module and that you also helped me to configure the gwiot7941e RFID reader library correctly.