arduino12/rdm6300 #14: rdm6300.update() Read the Card Multiple times

rezaalhadar, 2021-11-22:
How to prevent this library to read the card multiple times inside the loop.
I have this type of problem.

arduino12, 2021-11-22:
The RDM6300 module reads the card multiple times - as long as it is near it.. - and it sends the ID via its UART TX line.
All this lib is doing is parsing the UART TX data when available and when `rdm6300.update()` is called.

If you want to save power or somthing - you can power off the RDM6300 module using external transistor after you detect a valid card..
Or call `rdm6300.end()` - this will just ignore the UART TX data sent from the RDM6300 module..

rezaalhadar, 2021-12-08:
thanks i get it worked to read once at a time by calling `rdm6300.end()` inside the function.

closed 2021-12-08 by rezaalhadar