arduino12/rdm6300 #15: Update not working after standby for Arduino IoT

tickietackie, 2022-01-16:
I want to use the library to wake the Arduino from standby mode and read the tag value afterwards.
This works great when reading the first time, but does not working the following times because the id is always the same after then.

Any ideas how to solve this?

```
// **** INCLUDES *****
#include <rdm6300.h>
#include "LowPower.h"

// External interrupt on pin 0 (use pin 0 to 24, except pin 4 on Arduino Zero)
const int wake_pin = 9;
unsigned char count = 10;

Rdm6300 rdm6300;

void setup()
{
  // Wait for serial USB port to open
  while (!SerialUSB);
  SerialUSB.println("***** ATSAMD21 Standby Mode Example *****");

  Serial1.begin(RDM6300_BAUDRATE); // read the doc above, find the RX pin number in the link
  rdm6300.begin(&Serial1);

  pinMode(LED_BUILTIN, OUTPUT);
  digitalWrite(LED_BUILTIN, LOW);

  // ***** IMPORTANT *****
  // Delay is required to allow the USB interface to be active during
  // sketch upload process
  SerialUSB.println("Entering standby mode in:");
  for (count; count > 0; count--)
  {
    SerialUSB.print(count);
    SerialUSB.println(" s");
    delay(1000);
  }
  // *********************

  // External interrupt on pin (example: press of an active low button)
  // A pullup resistor is used to hold the signal high when no button press
  attachInterrupt(wake_pin, blink, LOW);
}

bool sleeping = true;

void loop()
{
  //if (sleeping == true) {
  SerialUSB.println("Entering standby mode.");
  SerialUSB.println("Apply low signal to wake the processor.");
  SerialUSB.println("Zzzz...");
  // Detach USB interface
  USBDevice.detach();
  // Enter standby mode
  LowPower.standby();

  // Attach USB interface
  USBDevice.attach();

  while (!Serial1.available()) {
    //wait
  }

  if (rdm6300.update()) {
    int tagId = rdm6300.get_tag_id();

    // wait for debug usb serial to be up and running
    // does not change the false behavior of the tag id updating, when turned off
    delay(2500);

    Serial.println("awake again");
    Serial.println(tagId);
    if (tagId == 5056730) {
      Serial.println("The tag was verified!");
      digitalWrite(LED_BUILTIN, HIGH);
    }
    else {
      Serial.println("False tag. Please use the one with id 5056730.");
    }
  }

  delay(3000);
  digitalWrite(LED_BUILTIN, LOW);
  sleeping = false;
  delay(10);
}

void blink(void)
{

}
```

Here I hold to different RFID tags in front of the sensor, but as you can see the tag id stays the same.

[image]

arduino12, 2022-01-16:
**Hi, @tickietackie **

**1.** What MCU board are you using?

**2.** Did you test the RDM6300 power consumption - how much current on IDLE / CARD-NEAR?
(I think it will be much bigger than the MCU- unless you power off the RDM6300)

**3.** My guess is that the bug occurred inside the UART peripheral / driver-
after wake-up from sleep maybe some registers / timing / cyclic buffer gets wrong..

**4.** I suggest that the wake-up pin will cause the MCU to restart like the RST pin -
this way - the UART and everything will be initialize and I think the RFID reading will be correct.

tickietackie, 2022-01-16:
Thanks for the fast reply.

1. Sorry just wrote it in the title, I am using an Arduino Nano 33 IoT, which features the SAMD21G18A
2. No, I did not so far, good point. I will check this. So you're implying this is no good solution anyway. Any idea how to optimise this?
3. Could be, that probably would be hard to fix myself.
4. This could be a workaround, but of course this will need a bit of time, which is not that great when waking up the device, also because at the setup a lot of other stuff is going on in the project, like connecting to Wi-Fi.

arduino12, 2022-01-16:
I don't have a Nano 33 IoT, but it looks nice!
I see many other chips on it (IMU, u-blocks, ATECC608A...) so you will have to put them all in low-power mode as well..

About 4, the MCU can know the reset-reason was something like PIN6 interrupt
-so it only runs the RFID code and not the WiFi code

Keep tring using the other examples here (HW serial, SW serial...) and report :)
You can also debug and just print the raw serial data using `Serial.read()` as hex-
this will tell us if the problem is inside the serial driver or in this lib (maybe after wake-up all of its internal variables goes wild)

arduino12, 2022-01-16:
Did you see the **[previous issue](https://github.com/arduino12/rdm6300/issues/14#issuecomment-975922771)**?
Maybe this will solve it:
```
rdm6300.end();
LowPower.standby(); // Enter standby mode
Serial1.begin(RDM6300_BAUDRATE);
rdm6300.begin(&Serial1);
```

Let us know..!

tickietackie, 2022-01-16:
> Did you see the **[previous issue](https://github.com/arduino12/rdm6300/issues/14#issuecomment-975922771)**? Maybe this will solve it:
> 
> ```
> rdm6300.end();
> LowPower.standby(); // Enter standby mode
> Serial1.begin(RDM6300_BAUDRATE);
> rdm6300.begin(&Serial1);
> ```
> 
> Let us know..!

This unfortunately does not solve the behaviour. There is no change.
I will try reading the Serial raw data and report back.

tickietackie, 2022-01-16:
SO this really looks like a timing issues in some way. 
With exactly this code it sometimes worked, when using 100 ms as a delay there.

```
// **** INCLUDES *****
#include <rdm6300.h>
#include "LowPower.h"

// External interrupt on pin 0 (use pin 0 to 24, except pin 4 on Arduino Zero)
const int wake_pin = 9;
unsigned char count = 10;

Rdm6300 rdm6300;

void setup()
{
  // Wait for serial USB port to open
  while (!SerialUSB);
  SerialUSB.println("***** ATSAMD21 Standby Mode Example *****");

  Serial1.begin(RDM6300_BAUDRATE); // read the doc above, find the RX pin number in the link
  rdm6300.begin(&Serial1);

  pinMode(LED_BUILTIN, OUTPUT);
  digitalWrite(LED_BUILTIN, LOW);

  // ***** IMPORTANT *****
  // Delay is required to allow the USB interface to be active during
  // sketch upload process
  SerialUSB.println("Entering standby mode in:");
  for (count; count > 0; count--)
  {
    SerialUSB.print(count);
    SerialUSB.println(" s");
    delay(1000);
  }
  // *********************

  // External interrupt on pin (example: press of an active low button)
  // A pullup resistor is used to hold the signal high when no button press
  attachInterrupt(wake_pin, blink, LOW);
}

bool sleeping = true;
int incomingByte = 0; // for incoming serial data

void loop()
{
  //if (sleeping == true) {
  SerialUSB.println("Entering standby mode.");
  SerialUSB.println("Apply low signal to wake the processor.");
  SerialUSB.println("Zzzz...");
  //rdm6300.end();
  // Detach USB interface
  USBDevice.detach();
  // Enter standby mode
  LowPower.standby();

  // Attach USB interface
  USBDevice.attach();

  int data[1000];;

  int i = 0;

  delay(100);  //when using 200 or sth else here, the output will be something different
  
  while (!Serial1.available()) {
  }

//saving raw serial data
  while (Serial1.available()) {
    incomingByte = Serial1.read();
    data[i] = incomingByte;
    i++;
  }

  delay(2500);

  Serial.println("Raw data read from serial1:");
  for (int x = 0; x < i; x++)
  {
    Serial.print(data[x], HEX);
  }

  if (rdm6300.update()) {
    int tagId = rdm6300.get_tag_id();

    // wait for debug usb serial to be up and running
    // does not change the false behavior of the tag id updating, when turned off
    delay(2500);

    Serial.println("awake again");
    Serial.println(tagId);
    if (tagId == 5056730) { //=4D28DA,  //43BF6B == 4439915
      Serial.println("The tag was verified!");
      digitalWrite(LED_BUILTIN, HIGH);
    }
    else {
      Serial.println("False tag. Please use the one with id 5056730.");
    }
  }

  delay(3000);
  digitalWrite(LED_BUILTIN, LOW);
  sleeping = false;
  delay(10);
}

void blink(void)
{

}
```

[image]

I tried using 200ms as delay as well, which lead to the following raw output, which I don't really know how to interpret.
I changed the RFID Tags so the first output is Tag 1, second Tag 2, then Tag 1 again and at last Tag2.

[image]

tickietackie, 2022-01-17:
So, I partially solved it with a workaround. 
This can be improved for sure but for now its the best I can come up with.
I simply let the Arduino do some cycles before going to sleep again, which then results in successful reading.

```
// **** INCLUDES *****
#include <rdm6300.h>
#include "LowPower.h"

// External interrupt on pin 0 (use pin 0 to 24, except pin 4 on Arduino Zero)
const int wake_pin = 9;
unsigned char count = 10;

Rdm6300 rdm6300;

void setup()
{
  // Wait for serial USB port to open
  while (!SerialUSB);
  SerialUSB.println("***** ATSAMD21 Standby Mode Example *****");

  Serial1.begin(RDM6300_BAUDRATE); // read the doc above, find the RX pin number in the link
  rdm6300.begin(&Serial1);

  pinMode(LED_BUILTIN, OUTPUT);
  digitalWrite(LED_BUILTIN, LOW);

  // ***** IMPORTANT *****
  // Delay is required to allow the USB interface to be active during
  // sketch upload process
  SerialUSB.println("Entering standby mode in:");
  for (count; count > 0; count--)
  {
    SerialUSB.print(count);
    SerialUSB.println(" s");
    delay(1000);
  }
  // *********************

  // External interrupt on pin (example: press of an active low button)
  // A pullup resistor is used to hold the signal high when no button press
  attachInterrupt(wake_pin, blink, LOW);
}

bool sleeping = true;
int minimumWaitCycles = 50;
int minimumWaitCyclesCount = 0;

void loop()
{
  if (sleeping == true) {
    SerialUSB.println("Entering standby mode.");
    SerialUSB.println("Apply low signal to wake the processor.");
    SerialUSB.println("Zzzz...");
    //rdm6300.end();
    // Detach USB interface
    USBDevice.detach();
    // Enter standby mode
    LowPower.standby();

    // Attach USB interface
    USBDevice.attach();

    minimumWaitCyclesCount = 0;

    //delay(400) for productive
    delay(2500);  //for debug
  }

  //wait for uart being available
  while (!Serial1) {
  }
  //when debugging, wait for usb port to come back
  while (!Serial) {
  }

  if (rdm6300.update()) {
    int tagId = rdm6300.get_tag_id();

    // wait for debug usb serial to be up and running
    // does not change the false behavior of the tag id updating, when turned off
    //delay(2500);

    Serial.println("awake again");
    Serial.println(tagId);
    if (tagId == 5056730) {
      Serial.println("The tag was verified!");
    }
    else {
      Serial.println("False tag. Please use the one with id 5056730.");
    }
  }

  digitalWrite(LED_BUILTIN, rdm6300.is_tag_near());
//do only sleep again, when tag is not near anymore and the Minimum cycles are cycled 
  if (rdm6300.is_tag_near() || minimumWaitCyclesCount < minimumWaitCycles) {
      sleeping = false;
  }
  else {
     sleeping = true;
  }

  minimumWaitCyclesCount++;
  delay(10);
}

void blink(void)
{

}
```

arduino12, 2022-01-19:
> Also SoftwareSerial is not available on the Arduino IoT. This is also the reason, why this library won't compile right away for the IoT. I had to remove all the SoftwareSerial stuff. The reason behind SoftwareSerial missing, are the hardware Serial features of the IoT. It offers hardware serials that can be assigned to different pins with I/O multiplexing.
> That's why I am using the hardware serial example.

So this need to be fixed **[here](https://github.com/arduino12/rdm6300/blob/master/src/rdm6300.h#L10-L23)**, need to find the equivalent of `ARDUINO_ARCH_ESP32` for the Nano 33 IoT and add it to there.
Please let me know if you find this so I can add it can you try with **[this](https://github.com/arduino/ArduinoCore-samd/blob/master/boards.txt#L248)**? :)

About the wake-up, try to make a minimum example - without the RDM6300 HW and SW,
Can try connect another simple arduino that just sens Serial data in a loop (and toggle the wake-up GPIO)-
If you see you are missing data after the wake-up - it means this bug need to be opened inside the **[official repo](https://github.com/arduino/ArduinoCore-samd)**.

tickietackie, 2022-01-19:
To the first remark I would suggest using ARDUINO_ARCH_SAMD to check for the SAMD architecture in general as I am pretty sure all of theses have no SoftwareSerial available, not the SAMD20 (e.g. Arduino Zero), nor SAMD21 like the IoT.

```
void setup() {
  Serial.begin(115200);
  while (!SerialUSB);
  delay(2000);
  
#if defined(ARDUINO_ARCH_SAMD)
  SerialUSB.println("I am SAMD");  //outputs true for SAMD MCU
#else
  SerialUSB.println("I am not SAMD");
#endif
}

void loop() {
  // put your main code here, to run repeatedly:
}
```

arduino12, 2022-02-09:
**Hi,**

I have just added support for the amazing SAMD family :)

On SAMD, the `Serial` objects comes from **[variant.cpp](https://github.com/arduino/ArduinoCore-samd/blob/master/variants/nano_33_iot/variant.cpp#L188)** and are instances of **[Uart](https://github.com/arduino/ArduinoCore-samd/blob/master/cores/arduino/Uart.h#L31)** class not HardwareSerial like in other MCUs..
Now the **[read_to_serial](https://github.com/arduino12/rdm6300/blob/master/examples/read_to_serial/read_to_serial.ino)** and the **[hardware_serial](https://github.com/arduino12/rdm6300/blob/master/examples/hardware_uart/hardware_uart.ino)** examples compiles out of the box.

@tickietackie can you please approve that they actually work on the hardware?

tickietackie, 2022-02-13:
Nice to hear that.
I could just test the newly released version and it compiles and works flawlessly. 
Thanks for the update :)

arduino12, 2022-02-13:
Great!

So the issue seems to be missing sleep cycles inside the `"LowPower.h"` lib - here is @tickietackie workaround:

> So, I partially solved it with a workaround. This can be improved for sure but for now its the best I can come up with. I simply let the Arduino do some cycles before going to sleep again, which [...]

closed 2022-02-13 by arduino12