arduino12/rdm6300 #17: Constantly outputting chipID

Paytrix, 2022-03-11:
Is there a way to use the library so that it constantly outputs the chipID if a tag is nearby? 

Can't get it to work. Would be awesome, if anybody had an example :)

arduino12, 2022-03-13:
**Hi,**

Please read the readme - last line is about `is_tag_near()`!

Here is an example that keeps printing the same tag-id as long as it is near the module:

```
void loop()
{
	static uint32_t tag_id = 0;

	/* if non-zero tag_id, update() returns true- a new tag is near! */
	if (rdm6300.update())
		tag_id = rdm6300.get_tag_id();

	if (rdm6300.is_tag_near())
		Serial.println(tag_id, HEX);

	delay(10);
}
```

Didn't tested so let us know if it worked for you?
(add the top part of the code from the [read_to_seria](https://github.com/arduino12/rdm6300/blob/master/examples/read_to_serial/read_to_serial.ino) example..)

Paytrix, 2022-03-17:
Hi, 

Thanks for your response! 

I actually did that for testing purposes and found out that you would have to be extremely carefull with holding the chip still and all. Not only that... if you have a little bit more going on on the ESP, because you often mant some more functionallity than just reading out a serial stream, you need to implement some sort of filtering of the `is_tag_near()` bool value.

So I just added a dirty counter, that filters out inconsistent `is_tag_near()` flickering.

```
if (rdm6300.update()) {
  t_chipID = rdm6300.get_tag_id();
}

if (rdm6300.is_tag_near()) {
  timeout = 0;
  chipID = t_chipID;
} else {
  timeout++;
  if (timeout > 10) {
    chipID = 0;
  }
}
```

arduino12, 2022-04-02:
**Hi,**

I played with the HW today and found out that the ```is_tag_near``` was [originally filtered befor a PR bug](https://github.com/arduino12/rdm6300/commit/076ae4c74adcc0fe121ab4717d47de65749b53fb)..
The RDM6300 module sends the tag info every ~65ms when it is near it:
[image]

I made some API changes and bump the major version to [v2.0.0](https://github.com/arduino12/rdm6300/releases/tag/v2.0.0)!
Here is an updated example that keeps printing the same tag_id as long as it is near the module:
```
void loop()
{
	uint32_t tag_id = rdm6300.get_tag_id(); // use get_new_tag_id() for single print per "new" tag sensed

	if (tag_id)
		Serial.println(tag_id , HEX);

	delay(10);
}
```

closed 2022-04-05 by Paytrix

Paytrix, 2022-04-05:
Hi,

awesome! Thanks for the update! 

Is the 10ms delay still necessary, or just for not overflowing the serial port?

arduino12, 2022-04-06:
The delay is just for the serial,
this works without it (tested on Arduino nano):
```
void loop()
{
	Serial.print(rdm6300.get_tag_id() ? '1' : '0');
}
```