arduino12/rdm6300 #18: RDM6300 with ESP8266 (D1 Mini)

diarmuidr3d, 2022-05-07:
Hi there,

I'm a little stuck following the setup for this, wondering if I'm encountering a bug or maybe just newbie idiocy.

I've wired up the RDM6300 with the ESP8266 (D1 Mini) as in the diagram from the guide on the github repo which also matches the code example from the repo here [examples/esp8266_hardware_uart](https://github.com/arduino12/rdm6300/blob/master/examples/esp8266_hardware_uart/esp8266_hardware_uart.ino).
RDM6300 5v -> Board 5v
RDM6300 GND -> Board GND
RDM6300 TX -> Board D7
Board TX -> Board D4

When I tried using the Arduinio IDE's Serial monitor I just get a massive unprocessable amount of "⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮x⸮" printed over and over. (I looked at #1 and my Baud rate is matching the code).
To try to debug another way, I added code to blink the LED if a new tag is found in the example, and it just blinks every second, regardless of whether there's a nearby tag or not.

Have I encountered an issue or just done something wrong?

Code I'm running is below (mostly copied from the example, just the LED blinking added):
`#include <rdm6300.h>

#define RDM6300_RX_PIN 13 // can be only 13 - on esp8266 force hardware uart!
#define READ_LED_PIN 16

Rdm6300 rdm6300;

void setup()
{
  /* Serial1 is the debug! remember to bridge GPIO-01 to GPIO-02 */
  Serial1.begin(115200);

  pinMode(READ_LED_PIN, OUTPUT);
  digitalWrite(READ_LED_PIN, LOW);

  rdm6300.begin(RDM6300_RX_PIN);
  pinMode(LED_BUILTIN, OUTPUT);

  Serial1.println("\nPlace RFID tag near the rdm6300...");
}

void loop()
{
  /* get_new_tag_id returns the tag_id of a "new" near tag,
  following calls will return 0 as long as the same tag is kept near. */
  if (rdm6300.get_new_tag_id())
    Serial1.println(rdm6300.get_tag_id(), HEX);
    digitalWrite(LED_BUILTIN, HIGH);
    delay(1000);
    digitalWrite(LED_BUILTIN, LOW);

  /* get_tag_id returns the tag_id as long as it is near, 0 otherwise. */
  digitalWrite(READ_LED_PIN, rdm6300.get_tag_id());

  delay(10);
}`

arduino12, 2022-05-07:
**Hi,**

The original example code controls turn on an LED when a RFID tag is near- **[see line 31](https://github.com/arduino12/rdm6300/blob/master/examples/esp8266_hardware_uart/esp8266_hardware_uart.ino#L31)**,
I made it for NodeMCU that **[has LED on GPIO16](https://lowvoltage.github.io/2017/07/09/Onboard-LEDs-NodeMCU-Got-Two)**:
[image]

The [`LED_BUILTIN` on the d1_mini board is GPIO2 (D4)](https://github.com/esp8266/Arduino/blob/master/variants/d1_mini/pins_arduino.h#L35) - **don't use it because this GPIO is used as UART!**

Better just connect another LED and a resistor via GPIO16 (D0) to 3V...

Your added code is not needed because of the above,
also in C and C++ a block of code must be inside **curly brackets**, it should be:
```
  if (rdm6300.get_new_tag_id()) {
    Serial1.println(rdm6300.get_tag_id(), HEX);
    digitalWrite(LED_BUILTIN, HIGH);
    delay(1000);
    digitalWrite(LED_BUILTIN, LOW);
  }
```

Make sure you connected all pins well - because I just connected my d1_mini to the RDM6300 **and it works!**

[image]

RDM6300 5V -> Board 5V
RDM6300 GND -> Board GND
RDM6300 TX -> Board D7
Board TX -> Board D4
Board D0 -> LED -> 470ohm resistor -> Board 3V

**1.** Does the LED turn off when a tag is near?
**2.** Does the text "Place RFID tag near the rdm6300..." appears on the serial monitor (115200bps) when you press the reset button?
**3.** Does it works with the [basic software-serial example](https://github.com/arduino12/rdm6300/blob/master/examples/read_to_serial/read_to_serial.ino)?

diarmuidr3d, 2022-05-09:
Hi @arduino12, thanks for the detailed response, really appreciate the effort put into that!
Sorry, I must just have been using dodgy cables, I rewired it with all new cables and I see the message "Place the RFID tag..." now when I hit the reset button.
Thanks so much for your help!
If you have a "Buy me a coffee" page or a github sponsor would love to pay for a drink for you, thanks!

arduino12, 2022-05-10:
Happy to hear that and thanks for your support!

closed 2022-05-10 by arduino12