arduino12/rdm6300 #19: tag read by the library is to short

MarkusPL0, 2023-06-27:
I've got RFID card which HEX number is 330092A6D7 (this number gave me a women, who generates cards - she sees such a number in their system). 
My problem is, that RDM6300 library returns only part of this number - namely 92A6D7.  Just for testing  I've connected directly my laptop with rdm6300 module (via USB/TTL cable). When talking directly with the rdm6300 module I receive 330092A6D7D0  on the serial monitor.  It's the same number which I got + D0 at the end. 

How can I modify this library to get full  RFID number?
thx in advance!
Markus.

MarkusPL0, 2023-06-27:
Hi, thx for your instant answer! 

I think the assumption that all 40bit tags begins with 33 is wrong! 
I've got 5 RFID cards with me that begins with 61 and one keychain which begins with (3800928ee4). 
Maybe some vendors ignore standard  and use all 40 bits for ID. I don't know. In the meantime (I didn't expect you're gonna answer so fast :)  I tested another library:
https://github.com/arliones/RDM6300-Arduino

This one can handle all bits  -  the  description:

"Driver for RDM6300 RFID reader to Arduino. RDM6300 supports 125KHz tags that store 40-bit IDs. The driver handles the IDs as 64-bit integers (unsigned long long), keeping the 24 MSB always in zero."

tomorrow at work I'm gonna check your fix (changing to uint64_t)

thx!

MarkusPL0, 2023-06-28:
Hi,
unfortunately changing uint32_t to uint64_t did not help ; there must be something else

arduino12, 2023-06-28:
**Hi,**

Well it seem that those 125kHz tags do use 40bits! 😮

For backward-compatibility I won't change the library because many already use it with 32bits.
(If I knew it back then - I would have use all 40bits..)

**For 40bits support you need to:**
1. Change all tag_id related functions and variables from `uint32_t` to `uint64_t` in `rdm6300.h` and `rdm6300.cpp` files.
2. Change [**these lines**](https://github.com/arduino12/rdm6300/blob/master/src/rdm6300.cpp#L86L92) to store the extra 8bits into the tag_id:
```
	/* add the first 8bits of tag_id so we get 40bits tag_id */
	buff[3] = 0;
	tag_id |= (uint64_t)strtol(buff + 1, NULL, 16) << 32;

	/* xor the tag_id and validate checksum */
	for (uint8_t i = 0; i < 40; i += 8)
		checksum ^= ((tag_id >> i) & 0xFF);
```

Didn't tested yet, you are welcome to try and write if it worked for you so others can use it :)

MarkusPL0, 2023-06-28:
Hi,
now it works fine!  

take a look at examples below. the difference is clear :) 
92a6d7-> 330092a6d7
928ee4 -> 3800928ee4
5c1f3d -> 61005c1f3d
thanks  a lot!
PS
maybe it would be good idea to define the bit length in rdm6300.h  (with default value of 32) and handle accordingly in rdm6300.cpp?

arduino12, 2023-06-28:
**Hi,**

**Thanks for testing!**
I hope to find some time next month to add that option to go 40bit and test it on my hardware before pushing the changes🙃

Will close this issue once its done. 👍

mikekgr, 2023-12-16:
> tag_id |= (uint64_t)strtol(buff + 1, NULL, 16) << 32;

Dear Sirs,
as far as I understood, the above changes should be done to the source files rdm6300.h  and  rdm6300.cpp 
I tried to implement the changes but it was unsuccessful so please can someone give me these modified files ?
Just to mention that I am planning to use the library on my Arduino Uno (I state that because of the uint64 that I am not sure that it is support it).
Many thanks

mikekgr, 2023-12-18:
> Hi, now it works fine!
> 
> take a look at examples below. the difference is clear :) 92a6d7-> 330092a6d7 928ee4 -> 3800928ee4 5c1f3d -> 61005c1f3d thanks a lot! PS maybe it would be good idea to define the bit length in rdm6300.h (with default value of 32) and handle accordingly in rdm6300.cpp?

@MarkusPL0 can you help me on this? I just want the modified files rdm6300.cpp   and   rdm6300.h  as I understood you have them already modified and working, so please help me. Thanks

MarkusPL0, 2023-12-19:
> @MarkusPL0 can you help me on this? I just want the modified files rdm6300.cpp and rdm6300.h 
> as I understood you have them already modified and working, so please help me. Thanks

sure
[rdm6300_modified.zip](https://github.com/arduino12/rdm6300/files/13714849/rdm6300_modified.zip)

mikekgr, 2023-12-19:
> > @MarkusPL0 can you help me on this? I just want the modified files rdm6300.cpp and rdm6300.h
> > as I understood you have them already modified and working, so please help me. Thanks
> 
> sure [rdm6300_modified.zip](https://github.com/arduino12/rdm6300/files/13714849/rdm6300_modified.zip)

@MarkusPL0  Many thanks for your kind help.
Best Regards

mikekgr, 2023-12-19:
> > @MarkusPL0 can you help me on this? I just want the modified files rdm6300.cpp and rdm6300.h
> > as I understood you have them already modified and working, so please help me. Thanks
> 
> sure [rdm6300_modified.zip](https://github.com/arduino12/rdm6300/files/13714849/rdm6300_modified.zip)

Dear @MarkusPL0  ,
I overwrite the files library src files from your rdm6300_modified.zip and I tried the simplest example read_to_serial.ino now when I try to compile it (Arduino: 1.8.19 (Windows 10), Board: "Arduino Uno"), I got error in the line 41: Serial.println(rdm6300.get_tag_id(), HEX);   
The error says: 
read_to_serial:41:43: error: call of overloaded 'println(uint64_t, int)' is ambiguous
   Serial.println(rdm6300.get_tag_id(), HEX);

What I did wrong  or  what I have to change in the line or sketch?

Thanks

MarkusPL0, 2023-12-20:
have you tried casting it directly to String   ->  String(rdm6300.get_tag_id(), HEX)

mikekgr, 2023-12-20:
Dear @MarkusPL0  ,
I first tester the below unmodified sketch (from the library's examples: read_to_serial), after using the modified library from your supplied zip 
`
#include <Arduino.h>
#include <rdm6300.h>

#define RDM6300_RX_PIN 4 // read the SoftwareSerial doc above! may need to change this pin to 10...
#define READ_LED_PIN 13

Rdm6300 rdm6300;

void setup()
{
	Serial.begin(115200);

	pinMode(READ_LED_PIN, OUTPUT);
	digitalWrite(READ_LED_PIN, LOW);

	rdm6300.begin(RDM6300_RX_PIN);

	Serial.println("\nPlace RFID tag near the rdm6300...");
}

void loop()
{
	/* get_new_tag_id returns the tag_id of a "new" near tag,
	following calls will return 0 as long as the same tag is kept near. */
	if (rdm6300.get_new_tag_id())
		Serial.println(rdm6300.get_tag_id(), HEX);

	/* get_tag_id returns the tag_id as long as it is near, 0 otherwise. */
	digitalWrite(READ_LED_PIN, rdm6300.get_tag_id());

	delay(10);
}
`
then according to your suggestion above, I modified the line from:
`
	if (rdm6300.get_new_tag_id())
		Serial.println(rdm6300.get_tag_id(), HEX);

`
to:

`
	if (rdm6300.get_new_tag_id())
		Serial.println(String(rdm6300.get_tag_id(), HEX));

`
but I got the folowing error:
C:\Users\Tester\AppData\Local\Temp\arduino_modified_sketch_25707\read_to_serial.ino: In function 'void loop()':
read_to_serial:41:50: error: call of overloaded 'String(uint64_t, int)' is ambiguous
   Serial.println(String(rdm6300.get_tag_id(), HEX));

call of overloaded 'String(uint64_t, int)' is ambiguous

Sorry but I can't find what is wrong, if you can please help me. 
Thanks

MarkusPL0, 2023-12-20:
I was testing it using Wemos D1 board (based on ESP8266). When you change your board to Wemos D1 (or whatever based on ESP8266) this program will compile without problems.  On Arduino UNO Board in fact there is  an error 

it looks like   Arduino Print class does not support 64-bit integers.  Have a look here, there is a workaround for that

https://github.com/greiman/SdFat/issues/130

mikekgr, 2023-12-20:
> I was testing it using Wemos D1 board (based on ESP8266). When you change your board to Wemos D1 (or whatever based on ESP8266) this program will compile without problems. On Arduino UNO Board in fact there is an error
> 
> it looks like Arduino Print class does not support 64-bit integers. Have a look here, there is a workaround for that
> 
> [greiman/SdFat#130](https://github.com/greiman/SdFat/issues/130)

Ok now it's clear that there are some limitations in the Arduino Uno and the related.
Looking for a workaround I found also a good library that's working fine with Arduino Uno, the library is:  https://github.com/RobTillaart/printHelpers/

Again, I want to thank you for your kind support.

Best Regards,
Mike Kranidis